veripool/verilog-perl #1663: Verilog-perl Parser test fails

ashfak12, 2020-03-21:
Few of Make file test included with verilog-perl (**Verilog-Perl-3.470**) fails. It seems like most of these failures are related to parser errors. Below is a snippet of failing report.

**Commands used in order:** 
1) perl Makefile.PL
2) make
3) make test

```
PERL_DL_NONLAZY=1 /usr/em64t_SLES11/pkgs/perl/5.14.1/bin/perl "-MExtUtils::Command::MM" "-e" "test_harness(0, 'blib/lib', 'blib/arch')" t/*.t
t/00_pod.t ............ ok
t/01_manifest.t ....... ok
t/02_help.t ........... ok
t/03_spaces.t ......... ok
t/04_critic.t ......... ok
t/05_yaml.t ........... ok
t/10_keywords.t ....... ok
t/12_splitbus.t ....... ok
t/14_numbers.t ........ ok
t/16_std.t ............ ok
t/20_getopt.t ......... ok
t/30_preproc.t ........ 
Dubious, test returned 25 (wstat 6400, 0x1900)
Failed 16/19 subtests 
t/32_noinc.t .......... 
Dubious, test returned 255 (wstat 65280, 0xff00)
Failed 2/7 subtests 
t/33_gzip.t ........... 
Dubious, test returned 255 (wstat 65280, 0xff00)
Failed 1/4 subtests 
t/34_parser.t ......... 
Dubious, test returned 2 (wstat 512, 0x200)
Failed 3/7 subtests 
t/35_sigparser.t ...... 
Dubious, test returned 2 (wstat 512, 0x200)
Failed 5/6 subtests 
t/36_sigmany.t ........ 
Dubious, test returned 2 (wstat 512, 0x200)
Failed 1/3 subtests 
	(less 1 skipped subtest: 1 okay)
t/40_netlist.t ........ 
Dubious, test returned 2 (wstat 512, 0x200)
Failed 16/17 subtests 
t/41_example.t ........ 
Dubious, test returned 2 (wstat 512, 0x200)
Failed 2/3 subtests 
t/42_dumpcheck.t ...... 
Dubious, test returned 2 (wstat 512, 0x200)
Failed 16/17 subtests 
t/43_storable.t ....... 
Dubious, test returned 2 (wstat 512, 0x200)
Failed 3/4 subtests 
t/44_create.t ......... ok
t/46_link.t ........... 
Dubious, test returned 2 (wstat 512, 0x200)
Failed 1/2 subtests 
t/48_leak.t ........... 
Dubious, test returned 255 (wstat 65280, 0xff00)
Failed 2/2 subtests 
t/49_largeish.t ....... 
Dubious, test returned 255 (wstat 65280, 0xff00)
Failed 4/4 subtests 
t/50_vrename.t ........ 
Dubious, test returned 2 (wstat 512, 0x200)
Failed 6/6 subtests 
t/51_vrename_kwd.t .... 
Dubious, test returned 255 (wstat 65280, 0xff00)
Failed 6/6 subtests 
t/56_editfiles.t ...... ok
t/58_vsplitmodule.t ... ok
t/60_vpassert.t ....... 
Dubious, test returned 17 (wstat 4352, 0x1100)
Failed 6/6 subtests 
t/80_vppreproc.t ...... 
Dubious, test returned 15 (wstat 3840, 0xf00)
Failed 15/15 subtests 
t/85_vhier.t .......... 
Dubious, test returned 2 (wstat 512, 0x200)
Failed 17/17 subtests 
t/86_vhier_tick.t ..... 
Dubious, test returned 1 (wstat 256, 0x100)
Failed 1/2 subtests 
t/87_vhier_unicode.t .. 
Dubious, test returned 25 (wstat 6400, 0x1900)
Failed 2/3 subtests 

Test Summary Report
-------------------
t/30_preproc.t      (Wstat: 6400 Tests: 3 Failed: 0)
  Non-zero exit status: 25
  Parse errors: Bad plan.  You planned 19 tests but ran 3.
t/32_noinc.t        (Wstat: 65280 Tests: 6 Failed: 1)
  Failed test:  4
  Non-zero exit status: 255
  Parse errors: Bad plan.  You planned 7 tests but ran 6.
t/33_gzip.t         (Wstat: 65280 Tests: 3 Failed: 0)
  Non-zero exit status: 255
  Parse errors: Bad plan.  You planned 4 tests but ran 3.
t/34_parser.t       (Wstat: 512 Tests: 4 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 7 tests but ran 4.
t/35_sigparser.t    (Wstat: 512 Tests: 1 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 6 tests but ran 1.
t/36_sigmany.t      (Wstat: 512 Tests: 2 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 3 tests but ran 2.
t/40_netlist.t      (Wstat: 512 Tests: 1 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 17 tests but ran 1.
t/41_example.t      (Wstat: 512 Tests: 1 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 3 tests but ran 1.
t/42_dumpcheck.t    (Wstat: 512 Tests: 1 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 17 tests but ran 1.
t/43_storable.t     (Wstat: 512 Tests: 1 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 4 tests but ran 1.
t/46_link.t         (Wstat: 512 Tests: 1 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 2 tests but ran 1.
t/48_leak.t         (Wstat: 65280 Tests: 0 Failed: 0)
  Non-zero exit status: 255
  Parse errors: Bad plan.  You planned 2 tests but ran 0.
t/49_largeish.t     (Wstat: 65280 Tests: 0 Failed: 0)
  Non-zero exit status: 255
  Parse errors: Bad plan.  You planned 4 tests but ran 0.
t/50_vrename.t      (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 6 tests but ran 0.
t/51_vrename_kwd.t  (Wstat: 65280 Tests: 0 Failed: 0)
  Non-zero exit status: 255
  Parse errors: Bad plan.  You planned 6 tests but ran 0.
t/60_vpassert.t     (Wstat: 4352 Tests: 0 Failed: 0)
  Non-zero exit status: 17
  Parse errors: Bad plan.  You planned 6 tests but ran 0.
t/80_vppreproc.t    (Wstat: 3840 Tests: 15 Failed: 15)
  Failed tests:  1-15
  Non-zero exit status: 15
t/85_vhier.t        (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 17 tests but ran 0.
t/86_vhier_tick.t   (Wstat: 256 Tests: 2 Failed: 1)
  Failed test:  1
  Non-zero exit status: 1
t/87_vhier_unicode.t (Wstat: 6400 Tests: 1 Failed: 0)
  Non-zero exit status: 25
  Parse errors: Bad plan.  You planned 3 tests but ran 1.
Files=34, Tests=374,  2 wallclock secs ( 0.13 usr  0.02 sys +  1.88 cusr  0.23 csys =  2.26 CPU)
Result: FAIL
```

On running one of the failing test with verbose: **30_preproc.t**
```
make test TEST_VERBOSE=1 TEST_FILES=t/30_preproc.t 
make[1]: Entering directory `/Verilog-Perl-3.470/Parser'
make[1]: Leaving directory `/Verilog-Perl-3.470/Parser'
make[1]: Entering directory `/Verilog-Perl-3.470/Preproc'
make[1]: Leaving directory `/Verilog-Perl-3.470/Preproc'
PERL_DL_NONLAZY=1 /usr/em64t_SLES11/pkgs/perl/5.14.1/bin/perl "-MExtUtils::Command::MM" "-e" "test_harness(1, 'blib/lib', 'blib/arch')" t/30_preproc.t
t/30_preproc.t .. 
1..19
ok 1 - use Verilog::Getopt
ok 2 - use Verilog::Preproc
ok 3 - new
%Error: verilog/inc_def09.v:1: Expecting define name. Found: DEFINE

Stopped at t/30_preproc.t line 77
# Looks like you planned 19 tests but ran 3.
# Looks like your test exited with 25 just after 3.
Dubious, test returned 25 (wstat 6400, 0x1900)
Failed 16/19 subtests 

Test Summary Report
-------------------
t/30_preproc.t (Wstat: 6400 Tests: 3 Failed: 0)
  Non-zero exit status: 25
  Parse errors: Bad plan.  You planned 19 tests but ran 3.
Files=1, Tests=3,  0 wallclock secs ( 0.02 usr  0.00 sys +  0.02 cusr  0.00 csys =  0.04 CPU)
Result: FAIL
Failed 1/1 test programs. 0/3 subtests failed.
make: *** [test_dynamic] Error 25
```

wsnyder, 2020-03-22:
Sorry you are having trouble.

The tests failing are all of the tests relying on the C code, which seems more than suspicious.

According to CPAN testers it is all green: http://matrix.cpantesters.org/?author=wsnyder#Verilog-Perl-3.470 

So, is there something unusual about your system e.g. non-Linux based?  What distro is this?

Can you please attach the build log when you made Verilog-Perl?

wsnyder, 2020-03-22:
Also your version of flex and bison please.

ashfak12, 2020-03-22:
I am using the following Linux distribution: SUSE Linux Enterprise Server 11 (x86_64)

Perl Version: 5.14.1
Gcc version: 4.7.2
Flex version: 2.5.4
Bison version: 2.4

I've tried older older Verilog-Perl releases too, and face the same issue. (Verilog-Perl-3.466 & Verilog-Perl-3.468). CSPAN testers for all these versions are green too.

Here is my make compile log.
[build.log](https://github.com/veripool/verilog-perl/files/4364535/build.log)

I also un-commented the following line in 30_preproc.t to print more debug info:
**Line 68:** `$pp->debug(9);`

Here is the test log with this debug set. 
[30_preproc.log](https://github.com/veripool/verilog-perl/files/4364551/30_preproc.log)

Here inc.out file from **30_preproc** test, if that helps..
[inc.txt](https://github.com/veripool/verilog-perl/files/4364542/inc.txt)

wsnyder, 2020-03-22:
Bison 2.4 is likely the problem as is over 12 years old.  Use at least 2.4.1.

wsnyder, 2020-03-22:
Bison 2.4.1 and flex 2.5.35 are the oldest known to work.

ashfak12, 2020-03-22:
I updated to Bison 3.0.4 & Flex 2.6.4. Now most of tests are passing with **Verilog-Perl-3.470**

Two tests are still failing with following errors

```
t/51_vrename_kwd.t .... ok   
t/56_editfiles.t ...... ok   
t/58_vsplitmodule.t ... ok   
t/60_vpassert.t ....... 1/6 %Warning: Line 69 mismatches; test_dir/.vpassertcall/example.v t/60_vpassert.out
GOT: `line 29 "verilog/example.v" 0
EXP: `line 36 "verilog/example.v" 0
To update reference: HARNESS_UPDATE_GOLDEN=1 t/60_vpassert.t

#   Failed test 'diff'
#   at t/60_vpassert.t line 36.
# Looks like you failed 1 test of 6.
t/60_vpassert.t ....... Dubious, test returned 1 (wstat 256, 0x100)
Failed 1/6 subtests 
        (less 1 skipped subtest: 4 okay)
t/80_vppreproc.t ...... ok     
t/85_vhier.t .......... 9/17 %Warning: Line 4 mismatches; test_dir/vhier.out t/85_vhier_skiplist.out
GOT:      |--subsub0 v_hier_subsub
EXP: *EOF*
To update reference: HARNESS_UPDATE_GOLDEN=1 t/85_vhier.t

#   Failed test 'vhier file compare'
#   at t/85_vhier.t line 36.
# Looks like you failed 1 test of 17.
t/85_vhier.t .......... Dubious, test returned 1 (wstat 256, 0x100)
Failed 1/17 subtests 
t/86_vhier_tick.t ..... ok   
t/87_vhier_unicode.t .. ok   

Test Summary Report
-------------------
t/60_vpassert.t     (Wstat: 256 Tests: 6 Failed: 1)
  Failed test:  5
  Non-zero exit status: 1
t/85_vhier.t        (Wstat: 256 Tests: 17 Failed: 1)
  Failed test:  14
  Non-zero exit status: 1
Files=34, Tests=482,  6 wallclock secs ( 0.15 usr  0.06 sys +  4.27 cusr  0.42 csys =  4.90 CPU)
Result: FAIL
Failed 2/34 test programs. 2/482 subtests failed.
make: *** [test_dynamic] Error 255
```

wsnyder, 2020-03-22:
You are on the same flex & bison as me then which is good.

If you'd be willing to debug these it would be great, the vpassert is some line number isn't tracked right, the vhier is it seems to be ignoring or mismatching on the 85_vhier_skiplist.dat file.

Also what does

    printenv | grep LANG

say?  Try setting it to C and see if that helps.

But if you want to move on it should be functional enough to force install and use.

ashfak12, 2020-03-23:
Here are few of my env varaibles

```
printenv | grep LANG
LANG=en_US.UTF-8

printenv | grep LC_
LC_NUMERIC=C
LC_MESSAGES=C
LC_CTYPE=C
LC_COLLATE=C
LC_MONETARY=C
LC_TIME=C
```
I tried changing LANG to C, but didn't see any difference in results.

I ran t/60_vpassert.t test, and found that the difference is just on one line.
**test_dir/.vpassertcall/example.v**(left) vs **t/60_vpassert.out** (right)
[image]

wsnyder, 2020-03-23:
Can you make sure the kit is correct (nothing modified) and no other versions of Verilog-Perl are installed?  Beyond that you'd need to turn on debug and start following the code between this and another machine to see why it's different.  The vhier test might be easier as it shouldn't be printing the line it is with the v_hier_subsub

ashfak12, 2020-03-23:
Yes I used the kit as is and I dont have any other versions of Verilog-Perl installed. 

I debugged 85_vhier.t test, and it seems like skiplist is read into array on vhier, but the grep expression to ignore it fails to match.

I modified **Line 322** on vhier from 
```perl
if (grep { $name =~ $_ } @Opt_Skiplist_Names) {
```
to 
```perl
if (grep (/^$name\b/, @Opt_Skiplist_Names)) {
```
With this change it passes the test. 

Using foreach loop the following code also works (replacing lines 322-325).
```perl
foreach (@Opt_Skiplist_Names) { 
  if ($_ =~ /^$name\s*$/) { 
    print "Debug: Skipping module $name" if $Debug;
    return;  
  }
}
```

Also closing the file handle before exiting on line 93 would be better. 
```perl
$fh->close();
```

wsnyder, 2020-03-23:
The skiplist contains regexps, which I think your proposed changes break.  What is strange to me is why the grep breaks on your system?

I'll add the close as certainly clearer, but it isn't technically needed as the exit will close files.

ashfak12, 2020-03-23:
I looked into it a bit more, and it seems that somehow my system is messing up while reading input from splitlist.dat file. Though i could see that it reads input fine (without any spaces in the end), if i add this substitution on each read line, before appending them to array, grep works fine.
**Line 379:**
```perl
    while (<$fh>) {
            chomp;
            $_ =~ s/\s+$//; #remove spaces
            push @Opt_Skiplist_Names, $_;
    }
```
Do you have any idea as to why this would be an issue?

wsnyder, 2020-03-24:
Check for DOS carriage returns. Maybe you unpacked on a windows system rather than linux?

ashfak12, 2020-03-24:
My bad. That was the issue. I unpacked in linux and all tests passes now. Thanks

closed 2020-03-24 by ashfak12