veripool/verilog-perl #1664: How to prevent link errors on parameterised module instances using vhier?

sjalloq, 2020-06-01:
Hi,

I'm trying to use vhier to build an XML tree of a design but I'm seeing errors for missing modules that are actually parameterised out.  You have the link_read_nonfatal option but I don't want to use this.  I want errors for missing modules that are used in the design.  What's the flow for using vhier when a module, mod_b, isn't in the tree?

` generate
    if PARAM
      mod_a u_a ();
    else
      mod_b u_b ();
    end
  endgenerate
`

wsnyder, 2020-06-01:
Sorry, but link_read_nonfatal is the only way to do this at present.  Since Verilog-Perl does not do elaboration, it can't know what modules are used or not used based on parameters.

Perhaps if your design uses mostly synthesis constructs you can use Verilator --xml output instead (see it's examples directory).

wsnyder, 2020-06-01:
Closing as answered, feel free to ask additional questions if needed.

closed 2020-06-01 by wsnyder