veripool/verilog-perl #1665: Preprocessor does not seem to handle import statement

PaulRolfe65, 2020-08-11:
The verilog preprocessor errors out when reading the following line:

  import <package_name>::*;

The error reported is:
 syntax error, unexpected IDENTIFIER, expecting PACKAGE-IDENTIFIER or STRING

The import looks to be legal & is handled correctly by other compilers

wsnyder, 2020-08-11:
Imports generally work.  Please attach a complete example that works on another tool, and shows the error. Thanks.

PaulRolfe65, 2020-08-12:
Here's a simple example

[package_fail.tar.gz](https://github.com/veripool/verilog-perl/files/5062210/package_fail.tar.gz)

I'm parsing the file test.sv. The include path is set to pick up the proj.vh file. 

Code fragment used:
   $pp = Verilog::Preproc->new(options=>$vlgOpts,
                               synthesis            =>(exists $opts{synthesis}       ? $opts{synthesis}       : 0),
                               include_open_nonfatal=>(exists $opts{includeOptional} ? $opts{includeOptional} : 0)
                              );
      $pp->open(filename=>$ppFile);
      # parse
      my $vlgParser = ASIC::File::Verilog::Parser->new($db,
         symbolTable => \@symTbl,
         files       => \%files,
         addLibCells => $addLibCells);
        $vlgParser->parse_preproc_file($pp);

I've also tried:
      while (defined ($_ = $pp->getline()))
      {
        $vlgParser->parse($_);
      }
      $vlgParser->eof();

But get the same error

This happens on the latest version (3.471)

The complete RTL compiles using Cadence Xcellium simulator

Many Thanks

wsnyder, 2020-08-12:
When I run

    vhier test_pkt.sv test.sv --forest

it works fine, so fundamentally there doesn't appear to be a problem with this package.

Your code snippit obviously isn't executable so I can't try it. I note it's calling your own package ASIC::File::Verilog::Parser, which could be doing any number of things wrong, so you'll need to dig into your code to see what is wrong.

PaulRolfe65, 2020-08-12:
Ahhh.....
vhier test_pkt.sv test.sv --forest
works for me too, BUT
vhier test.sv --forest
does not & I know that my code does not read the pkg file (I need to find someway of adding this package into the list of files read)

Many Thanks

wsnyder, 2020-08-12:
Glad we figured it out.

closed 2020-08-12 by wsnyder