veripool/verilog-perl #1667: `cpan install Verilog-Perl` seems to be broken

kevbroch, 2020-09-23:
I've done this many time in the past without issue but today it seems to be failing.  (I tried this on several systems and different versions of perl 5.30.0, perl 5.26.3.  All with same result).  I also tried variations of invoking cpan or cpanm like `perl -MCPAN -e 'install Verilog-Perl;'`

```
$ sudo cpan install Verilog-Perl
Loading internal logger. Log::Log4perl recommended for better logging
Reading '/root/.cpan/Metadata'
  Database was generated on Wed, 23 Sep 2020 15:17:03 GMT
>(error): Could not expand [Verilog-Perl]. Check the module name.
>(info): I can suggest names if you install one of Text::Levenshtein::XS, Text::Levenshtein::Damerau::XS, Text::Levenshtein, and Text::Levenshtein::Damerau::PP
>(info): and you provide the -x option on invocation.
>(error): Skipping Verilog-Perl because I couldn't find a matching namespace.

```

My workaround with `cpanm`:

```
$ sudo cpanm https://cpan.metacpan.org/authors/id/W/WS/WSNYDER/Verilog-Perl-3.470.tar.gz
--> Working on https://cpan.metacpan.org/authors/id/W/WS/WSNYDER/Verilog-Perl-3.470.tar.gz
Fetching https://cpan.metacpan.org/authors/id/W/WS/WSNYDER/Verilog-Perl-3.470.tar.gz ... OK
Configuring Verilog-Perl-3.470 ... OK
Building and testing Verilog-Perl-3.470 ... OK
Successfully installed Verilog-Perl-3.470
1 distribution installed
```

Anyone else see a problem?

lsteveol, 2021-02-22:
I see the same issue.

wsnyder, 2021-02-22:
CPAN doesn't seem to any longer understand the difference between the distro name and package name.  As a workaround use "cpan install Verilog::Language" instead.

closed 2021-02-22 by wsnyder