veripool/verilog-perl #1669: Parser didn't report macro with parsing library way

avendeng, 2021-02-01:
Hi, 

The macro defined report miss the part of the definition. And this part of definition is included by the Verilog file which is read by a compiler using "parsing library" way? Is the missing exist when the definition of the module is linked to a file or directory? If I use the command unexpected, someone may kindly correct me.

**Command**: /home/utils/perl-5.26/5.26.2-057/bin/vppreproc +libext+.vh+.v +incdir+{PWD}/ip {PWD}/ip/case.v -o vpp.out -dump-defines

**{PWD}/ip/case.v:**
module top;
module_a instance_module_a();
endmodule

**{PWD}/ip/LIB.v:**
module module_a;
`define MACRO 16
initial begin
$display("************************************************\n");
$display("In %m, Macro is : %d\n", `MACRO);
$display("************************************************\n");
end
endmodule

**Macro report: (missing MACRO)**
`define SV_COV_ASSERTION 20
`define SV_COV_CHECK 3
`define SV_COV_ERROR -1
`define SV_COV_FSM_STATE 21
`define SV_COV_HIER 11
`define SV_COV_MODULE 10
`define SV_COV_NOCOV 0
`define SV_COV_OK 1
`define SV_COV_OVERFLOW -2
`define SV_COV_PARTIAL 2
`define SV_COV_RESET 2
`define SV_COV_START 0
`define SV_COV_STATEMENT 22
`define SV_COV_STOP 1
`define SV_COV_TOGGLE 23

**VCS compile order:**
Parsing design file '../case.v'
Parsing library file '../LIB.v'
Top Level Modules:
       top

wsnyder, 2021-02-01:
I don't know what output you are expecting.
I don't know why you showed the VCS compile order, how does this matter?
But, your command line only shows case.v, so only defines in case.v will be shown, which seem to be none.
Note vppreproc is a **preprocessor** it will not resolve any library files, that is not the function of a preprocessor.

avendeng, 2021-02-01:
Thanks for your reply.

> I don't know what output you are expecting.
I  was a little confused with "+libext" and the library.  I expected to find out macro value defined in a library.
> I don't know why you showed the VCS compile order, how does this matter?
It should not be included here.
> But, your command line only shows case.v, so only defines in case.v will be shown, which seems to be none.
> Note vppreproc is a **preprocessor** it will not resolve any library files, that is not the function of a preprocessor.
Yeah, I see.  I am studying vppreproc now. May I confirm a question? The function of "preprocessor"  is to analyze the files, and no result is about simulation,  right?  It could not know the link result of simulation tools. This is my confusing point before your reply.

wsnyder, 2021-02-01:
Libext only describes how files are found - see the documentation.

A preprocessor only expands `defines and such, it does not simulate, see the IEEE specification.

closed 2021-02-01 by wsnyder