veripool/verilog-perl #1670: Can't load 'blib/arch/auto/Verilog/Parser/Parser.bundle' for module Verilog::Parser: dlopen(blib/arch/auto/Verilog/Parser/Parser.bundle, 0x0002): open("", O_RDONLY) failed with errno=2 at /System/Library/Perl/5.28/darwin-thread-multi-2level/DynaLoader.pm line 197

Shang0801, 2021-04-14:
When excute 
`cpan install Verilog::Language`

got a lot of " Can't load 'blib/arch/auto/Verilog/Parser/Parser.bundle' for module Verilog::Parser" error.

I have tried manual install and get the same issue when running make test.

Shang0801, 2021-04-14:
System: Apple Big Sur 
perl version: v5.28.2

wsnyder, 2021-04-14:
Can you please post more of the log? There should be more info before or after the message.

Shang0801, 2021-04-14:
Hi,
Yes. Here is the full log till the 1st error.
`Running make test
"/usr/bin/perl" -MExtUtils::Command::MM -e 'cp_nonempty' -- Parser.bs ../blib/arch/auto/Verilog/Parser/Parser.bs 644
Manifying 2 pod documents
"/usr/bin/perl" -MExtUtils::Command::MM -e 'cp_nonempty' -- Preproc.bs ../blib/arch/auto/Verilog/Preproc/Preproc.bs 644
Manifying 1 pod document
"/usr/bin/perl" -MExtUtils::Command::MM -e 'cp_nonempty' -- Parser.bs ../blib/arch/auto/Verilog/Parser/Parser.bs 644
No tests defined for Verilog::Parser extension.
"/usr/bin/perl" -MExtUtils::Command::MM -e 'cp_nonempty' -- Preproc.bs ../blib/arch/auto/Verilog/Preproc/Preproc.bs 644
No tests defined for Verilog::Preproc extension.
PERL_DL_NONLAZY=1 "/usr/bin/perl" "-MExtUtils::Command::MM" "-MTest::Harness" "-e" "undef *Test::Harness::Switches; test_harness(0, 'blib/lib', 'blib/arch')" t/*.t
t/00_pod.t ............ ok
t/01_manifest.t ....... ok

#   Failed test 'help result for: /usr/bin/perl -Iblib/arch -Iblib/lib -IPreproc/blib/arch -IPreproc/blib/lib blib/script/vhier --help 2>&1'
#   at t/02_help.t line 24.
#                   'Can't load 'blib/arch/auto/Verilog/Parser/Parser.bundle' for module Verilog::Parser: dlopen(blib/arch/auto/Verilog/Parser/Parser.bundle, 0x0002): open("", O_RDONLY) failed with errno=2 at /System/Library/Perl/5.28/darwin-thread-multi-2level/DynaLoader.pm line 197.
#   at blib/lib/Verilog/SigParser.pm line 11.
# Compilation failed in require at blib/lib/Verilog/SigParser.pm line 11.
# BEGIN failed--compilation aborted at blib/lib/Verilog/SigParser.pm line 11.
# Compilation failed in require at blib/lib/Verilog/Netlist/File.pm line 37.
# BEGIN failed--compilation aborted at blib/lib/Verilog/Netlist/File.pm line 37.
# Compilation failed in require at blib/lib/Verilog/Netlist.pm line 9.
# BEGIN failed--compilation aborted at blib/lib/Verilog/Netlist.pm line 9.
# Compilation failed in require at blib/script/vhier line 15.
# BEGIN failed--compilation aborted at blib/script/vhier line 15.
# '
#     doesn't match '(?^:--version)'

#   Failed test 'version result for: /usr/bin/perl -Iblib/arch -Iblib/lib -IPreproc/blib/arch -IPreproc/blib/lib blib/script/vhier --version 2>&1'
#   at t/02_help.t line 28.
#                   'Can't load 'blib/arch/auto/Verilog/Parser/Parser.bundle' for module Verilog::Parser: dlopen(blib/arch/auto/Verilog/Parser/Parser.bundle, 0x0002): open("", O_RDONLY) failed with errno=2 at /System/Library/Perl/5.28/darwin-thread-multi-2level/DynaLoader.pm line 197.
#   at blib/lib/Verilog/SigParser.pm line 11.
# Compilation failed in require at blib/lib/Verilog/SigParser.pm line 11.
# BEGIN failed--compilation aborted at blib/lib/Verilog/SigParser.pm line 11.
# Compilation failed in require at blib/lib/Verilog/Netlist/File.pm line 37.
# BEGIN failed--compilation aborted at blib/lib/Verilog/Netlist/File.pm line 37.
# Compilation failed in require at blib/lib/Verilog/Netlist.pm line 9.
# BEGIN failed--compilation aborted at blib/lib/Verilog/Netlist.pm line 9.
# Compilation failed in require at blib/script/vhier line 15.
# BEGIN failed--compilation aborted at blib/script/vhier line 15.
# '
#     doesn't match '(?^:Version.*[0-9])'

#   Failed test 'help result for: /usr/bin/perl -Iblib/arch -Iblib/lib -IPreproc/blib/arch -IPreproc/blib/lib blib/script/vpassert --help 2>&1'
#   at t/02_help.t line 24.
#                   'Can't load 'blib/arch/auto/Verilog/Parser/Parser.bundle' for module Verilog::Parser: dlopen(blib/arch/auto/Verilog/Parser/Parser.bundle, 0x0002): open("", O_RDONLY) failed with errno=2 at /System/Library/Perl/5.28/darwin-thread-multi-2level/DynaLoader.pm line 197.
#   at blib/script/vpassert line 20.
# Compilation failed in require at blib/script/vpassert line 20.
# BEGIN failed--compilation aborted at blib/script/vpassert line 20.
# '
#     doesn't match '(?^:--version)'

#   Failed test 'version result for: /usr/bin/perl -Iblib/arch -Iblib/lib -IPreproc/blib/arch -IPreproc/blib/lib blib/script/vpassert --version 2>&1'
#   at t/02_help.t line 28.
#                   'Can't load 'blib/arch/auto/Verilog/Parser/Parser.bundle' for module Verilog::Parser: dlopen(blib/arch/auto/Verilog/Parser/Parser.bundle, 0x0002): open("", O_RDONLY) failed with errno=2 at /System/Library/Perl/5.28/darwin-thread-multi-2level/DynaLoader.pm line 197.
#   at blib/script/vpassert line 20.
# Compilation failed in require at blib/script/vpassert line 20.
# BEGIN failed--compilation aborted at blib/script/vpassert line 20.
# '
#     doesn't match '(?^:Version.*[0-9])'

#   Failed test 'help result for: /usr/bin/perl -Iblib/arch -Iblib/lib -IPreproc/blib/arch -IPreproc/blib/lib blib/script/vppreproc --help 2>&1'
#   at t/02_help.t line 24.
#                   'Can't load 'blib/arch/auto/Verilog/Preproc/Preproc.bundle' for module Verilog::Preproc: dlopen(blib/arch/auto/Verilog/Preproc/Preproc.bundle, 0x0002): open("", O_RDONLY) failed with errno=2 at /System/Library/Perl/5.28/darwin-thread-multi-2level/DynaLoader.pm line 197.
#   at blib/script/vppreproc line 14.
# Compilation failed in require at blib/script/vppreproc line 14.
# BEGIN failed--compilation aborted at blib/script/vppreproc line 14.
# '
[... 26 lines cut ...]
# '
#     doesn't match '(?^:Version.*[0-9])'
# Looks like you failed 8 tests of 15.
t/02_help.t ........... 
Dubious, test returned 8 (wstat 2048, 0x800)
Failed 8/15 subtests 
	(less 2 skipped subtests: 5 okay)
t/03_spaces.t ......... ok
t/04_critic.t ......... ok
t/05_yaml.t ........... ok
[... cut ...]
Can't load 'blib/arch/auto/Verilog/Preproc/Preproc.bundle' for module Verilog::Preproc: dlopen(blib/arch/auto/Verilog/Preproc/Preproc.bundle, 0x0002): open("", O_RDONLY) failed with errno=2 at /System/Library/Perl/5.28/darwin-thread-multi-2level/DynaLoader.pm line 197.
  at t/30_preproc.t line 17.
Compilation failed in require at t/30_preproc.t line 17.
BEGIN failed--compilation aborted at t/30_preproc.t line 17.
# Looks like your test exited with 2 before it could output anything.
t/30_preproc.t ........ 
Dubious, test returned 2 (wstat 512, 0x200)
Failed 19/19 subtests 
`
Thanks!

wsnyder, 2021-04-14:
Not sure as it looks different from cases I've seen, but might simply be the build was inconsistent from an old version. Try deleting the whole area and rebuild.

closed 2025-05-15 by wsnyder