veripool/verilog-perl #1671: Verilog:Getopt doesn't recognize system environment variables in filelist

henry-hsieh, 2021-06-04:
When file list contains environment variables, it can't open the file. For example, following file list will cause the parser aborting.

```
${TOP}/fileA.v
${TOP}/fileB.v
${SUB}/file_sub.v
```

An easy fix is adding following code after [line 85](https://github.com/veripool/verilog-perl/blob/master/Getopt.pm#L85).
``` perl
$line =~ s/(?:^|\s)\/\/.*$//;
$line = `echo $line`
next if $line =~ /^\s*$/;
```

It lets system expand the environment variable and passes back to the `$line`.

wsnyder, 2021-06-04:
When iterating over the getopt arguments, call file_substitute($filename). This will substitute the env vars.

closed 2021-06-04 by wsnyder

henry-hsieh, 2021-06-05:
The problem occurs in vppreproc. When I load a filelist which contains environment variables, it can't recognize the path. Does vppreproc also provide file_substitute function, too?

henry-hsieh, 2021-06-05:
Hi @wsnyder ,

Thanks for your advisement. I found that the Getopt->parameter() will call Getopt->file_substitute() eventually to substitute the environment variables in filelist. Therefore, the parser shouldn't have the problem which I described originally.

So I jumped into the file_substitute() and saw some problems. The following code should fix it.

``` perl
sub file_substitute {
    my $self = shift;
    my $filename = shift;
    my $out = $filename;
    while ($filename =~ /\$([A-Za-z_0-9]+)\b/g) {
	my $var = $1;
	if (defined $ENV{$var}) {
	    $out =~ s/\$$var\b/$ENV{$var}/g;
	}
    }
    while ($filename =~ /\$\{([A-Za-z_0-9]+)\b\}/g) {
	my $var = $1;
	if (defined $ENV{$var}) {
	    $out =~ s/\$\{$var\b\}/$ENV{$var}/g;
	}
    }
    $out =~ s!^~!$ENV{HOME}/!;
    return $out;
}
```
1. The [regex](https://github.com/veripool/verilog-perl/blob/master/Getopt.pm#L390) for matching environment variables with curly brace, i.e, ${ABC}, is wrong. `$filename =~ /\$\{[A-Za-z_0-9]+\}\b/g` should be modified to `$filename =~ /\$\{([A-Za-z_0-9]+)\b\}/g`. You need parentheses to get `$1` in next line. Moreover, `\b` should be placed before `\}` because it's not a word.
2. The while loop should be separated. Or Perl can't loop over each match correctly.

wsnyder, 2021-06-05:
Agreed, thanks for debugging, changes pushed.  Please try the master version and if you need a new official release let me know.

henry-hsieh, 2021-06-05:
It would be great if you publish a new official release. Thanks!

wsnyder, 2021-06-06:
Released 3.478