veripool/verilog-perl #1672: Vppreproc: MMD feature

piecea, 2022-04-28:
Could the `-MMD` option (like that of gcc, but also of verilator) have a place in vppreproc? Note `–MMD` generates a dependency output file for all the user files. We are interested in this feature for automating our compile flow.

If vppreproc wants to focus only on preprocessing, maybe a -dump-includes option is more suitable?

Using only the verilog api, we have prototyped an implementation of the -MMD option and can share the code. but, we think it would be useful to be provide this option natively in vppreproc.

wsnyder, 2022-04-28:
I'd probably accept this if you want to make a pull request. 

I would note you might alternatively use "verilator -MMD -E" which should already support this.

wsnyder, 2025-05-15:
Closing due to age, as above might consider pull.

closed 2025-05-15 by wsnyder