veripool/verilog-perl #1677: need a parser for wire or reg signals

WilsonChen003, 2022-11-03:
Hi Wilson,
  This is greeting from another Wilson  ;-)
  I have a project for Verilog RTL stiching, also for IP/SOC integration, named as "HDLGen" --- https://github.com/WilsonChen003/HDLGen.
  But now I'm hitting a problem： how to parse any wire or reg logic ( not a whole design file) to know the signal name, and finally get its bit width? Sample Verilog code like below:

always @(
  cfg_is_int8_d0
  or res_a_00
  or res_b_00
  ) begin
    pp_in_l0_a_00 = cfg_is_int8_d0[0] ? {2'b0, res_a_00[31:16], 2'b0, res_a_00[15:0]} :
                    {4'b0, res_a_00[31:0]};
    pp_in_l0_b_00 = cfg_is_int8_d0[0] ? {2'b0, res_b_00[31:16], 2'b0, res_b_00[15:0]} :
                    {4'b0, res_b_00[31:0]};
end

  I have no good idea at all, as Verilog source coding style is various between different Engineers, so regular express may not work.
  But I think you'd cleaned such issue early before, so I open this issue as a help actually :).
  The purpose I need this function is that I want to achieve a goal in my tool:
    egineer just write logic code, the signals used in such logic can be automically defined as wire or reg with bit width correct --- this really help IP or ASIC enginneers.

Best Regards,
Wilson Chen

wsnyder, 2022-11-03:
Hi!

I'm not sure what is your question.  It sounds like you want reg/wires, if so see the test files the sources here and also https://github.com/veripool/verilog-perl#which-parser-package

If you want an example of how to edit code, see vppreproc here.

For your "stitch" function alternatively see https://github.com/veripool/verilog-mode AUTOs which have extensive industry usage.

closed 2022-11-03 by wsnyder

WilsonChen003, 2022-11-03:
Thanks Wilson for kindly help!
Yeah, EMAC is really cool! The AUTOPORT & AUTOREG is really what I want, but the auto instance is not, because I want to enable user to do his own conection through reg express.
I'll try to understand all these links to see if I can use some of these source code.
And thanks again.
Wilson Chen