veripool/verilog-perl #1678: Verilog::Preproc keeps comments containing word "module" although keep_comments=0

adrian1001, 2022-11-22:
If a line-comment in the verilog code contains the word "module", then Verilog::Preproc keeps "module" and the rest of line and returns it as if it was a code line, despite the fact that keep_comments=0.

Sample code:
```
use Verilog::Preproc;
my $file = "multiplier.sv";
my $pp = Verilog::Preproc->new(keep_comments=>0);
$pp->open(filename=>$file);
while (defined (my $line = $pp->getline())) {
   print $line;
}
```

Source "multiplier.sv":
```
// Description:
// This module implements a multiplier

module multiplier;
endmodule
```
Output:
```
module implements a multiplier

module multiplier;
endmodule
```
The correct output with all comments stripped should be:
```
module multiplier;
endmodule
```

wsnyder, 2022-11-22:
That's not possible.  The program you are using cannot match what you sent above, given that output.

adrian1001, 2022-11-22:
You are right: I just ran my own code stand-alone and the output is correct. I overlooked that my framework which is calling the very same code contains a pre-pre-processing and messes with the comments.

Sorry for the fuss!

closed 2022-11-22 by adrian1001