veripool/verilog-perl #1681: surpress include file can't open error

riggy2013, 2023-11-02:
Hi, 

I am using Verilog-perl as a lint. Is there any option to suppress the include file error such as:

%Error: ###.sv:##: Cannot open ###.svh
Stopped at .../Verilog/Parser.pm line 186.

thanks,

riggy2013.

wsnyder, 2023-11-02:
Use open_nonfatal=>1 when creating the Verilog::Preproc object.

Note Verilog-Perl is only in maintenance mode, depending on what you are doing, Verilator or another more complete parser might be better long term.

closed 2023-11-02 by wsnyder

riggy2013, 2023-11-03:
Hi, Wilson,

Yes, I add include_open_nonfatal=> in new Verilog::Netlist so it can run now.

It still prints bunches of %Error syntax errors for undefined macros which is quite common in single Verilog file. How can I suppress them too?

I use Verilog-Perl in my perl code so that I can add certain customized naming rules easily. I understand Verilator is a general program but not a callable package. What's your suggestion?

Thanks,
riggy

wsnyder, 2023-11-03:
There's not much you can do about those, the tool has no way of knowing what the define values might be.  You might want to look at [veriable ](https://github.com/chipsalliance/verible) which has a parser written to handle some incomplete files.

riggy2013, 2023-11-03:
I just wonder whether there is switch to turn such error message off. It seems no.

I found Verilog-Perl recognizes the interface name as cell name. for example.

`input intf_s sig_intf_a;`

it recognizes sig_intf_a as a cell name.

wsnyder, 2023-11-03:
Types need to by predeclared per IEEE in this parser. Verilog-Perl was not designed for standalone files that are not IEEE legal code.