veripool/verilog-perl #1683: vhier - Apply skiplist to input files and module files as well

mksoc, 2024-08-30:
As far as I understand, at the moment the `--skiplist` option in `vhier` is used only in conjunction with either `--cells` or `--forest`, but I think it would make sense to also skip modules defined in the skiplist when `--input-files` and `--module-files` are used, given that hierarchy is traversed with those options.

wsnyder, 2024-08-30:
You're perhaps correct, but that would require the parser which is a lower-level package to implement it.  And it might break old scripts.

At this time Verilog-Perl is in maintenance-only mode, so we're only taking bug fixes, so closing this out. Thanks for reporting though.

closed 2024-08-30 by wsnyder

mksoc, 2024-08-30:
Thanks for the reply and for all your good work!